adafruit/Adafruit_CircuitPython_PyPortal #11: Timezones cannot be an integer

makermelissa, 2019-03-06:
This is mainly a problem because the ESP32SPI secrets file uses the timezone as an integer which is causing PyPortal to crash. I noticed this when trying to use the OpenWeather example. Changing it to a string fixed it and I have submitted a PR to change it to a string by default, but it would still be nice to be able to use numbers.

ladyada, 2019-03-06:
time stuff needs a bit of reworking - we'll also be updating it to use adafruit.io instead of a free service, so we don't overload the nice people at worldtimeapi!

anecdata, 2019-03-07:
I think the issue with TZ offset numbers is that they're easier for people to remember sometimes than the strings, and definitely useful for concise _output_ of a specific date-time, but they're often imprecise for specifying your real timezone behavior... you might have to change your secrets.py twice a year for daylight savings time or other local complications. More detail than you'd ever want at: [https://data.iana.org/time-zones/theory.html](url)

ladyada, 2019-03-08:
we dont want to update for daylight savings, the goal here is to not have to do that - so i would prefer to stick to strings that fully determine what the time should be.

makermelissa, 2019-03-08:
Ok, it has been updated in ESP32SPI to use a string by default now, so I'm ok with closing this if we're not going that route.

ladyada, 2019-03-10:
thanks - this has been refactored for adafruit.io!

closed 2019-03-10 by ladyada