adafruit/Adafruit_CircuitPython_PyPortal #14: Call to Tilegrid fails after latest displayio update

jerryneedell, 2019-03-14:
see https://github.com/adafruit/circuitpython/issues/1646

ladyada, 2019-03-14:
yay - want to try a PR that try/excepts so either works?

jerryneedell, 2019-03-14:
Sure, but I can’t work on it until late today or tomorrow. Happy to do it then but if someone else wants to fix it first, go ahead.

jerryneedell, 2019-03-14:
:-(  display_text is also broken by this...
 https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/blob/master/adafruit_display_text/label.py#L107

I guess I should  do the same fix to it -- Her is what I was plannig -- let me know if is waht you want before I go too far down this rabbit hole.

```
diff --git a/adafruit_pyportal.py b/adafruit_pyportal.py
index 8e503bf..4bc9650 100644
--- a/adafruit_pyportal.py
+++ b/adafruit_pyportal.py
@@ -359,17 +359,27 @@ class PyPortal:
         if isinstance(file_or_color, str): # its a filenme:
             self._bg_file = open(file_or_color, "rb")
             background = displayio.OnDiskBitmap(self._bg_file)
-            self._bg_sprite = displayio.TileGrid(background,
-                                                 pixel_shader=displayio.ColorConverter(),
-                                                 position=position)
+            try:
+                self._bg_sprite = displayio.TileGrid(background,
+                                                     pixel_shader=displayio.ColorConverter(),
+                                                     position=position)
+            except TypeError:
+                self._bg_sprite = displayio.TileGrid(background,
+                                                     pixel_shader=displayio.ColorConverter(),
+                                                     x=position[0], y=position[1])
         elif isinstance(file_or_color, int):
             # Make a background color fill
             color_bitmap = displayio.Bitmap(320, 240, 1)
             color_palette = displayio.Palette(1)
             color_palette[0] = file_or_color
-            self._bg_sprite = displayio.TileGrid(color_bitmap,
-                                                 pixel_shader=color_palette,
-                                                 position=(0, 0))
+            try:
+                self._bg_sprite = displayio.TileGrid(color_bitmap,
+                                                     pixel_shader=color_palette,
+                                                     position=(0, 0))
+            except TypeError:
+                self._bg_sprite = displayio.TileGrid(color_bitmap,
+                                                     pixel_shader=color_palette,
+                                                     x=position[0], y=position[1])
         else:
             raise RuntimeError("Unknown type of background")
         self._bg_group.append(self._bg_sprite)
~
```

ladyada, 2019-03-14:
yep we have to go through and fix label to - just the same way. wanna do that and tag me on the PR? ill review!

jerryneedell, 2019-03-14:
sure

PR #15 (2019-03-15): fix calls to displayio.TileGrid usage of position
fixes #14
closed 2019-03-15 by kattni