adafruit/Adafruit_CircuitPython_PyPortal #16: Return UTC differential value from adafruit_pyportal.get_local_time()

CedarGroveStudios, 2019-03-18:
It would be very useful for calculating time-dependent values (such as sunrise and sunset times) if the `adafruit_pyportal.get_local_time` module also returned the UTC differential value for the local time zone.

ladyada, 2019-03-18:
that data *is* requested 
https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/master/adafruit_pyportal.py#L86
but we don't do anything with it
https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/master/adafruit_pyportal.py#L555
i dont think the python `time` module can track UTC - so would that be some other time struct? like where would we keep track of it :)

CedarGroveStudios, 2019-03-18:
I believe that in Python3, the base time is UTC and `localtime` is derived from that. Does CircuitPython do it that way or is the foundational time the local time? If that's the case, then perhaps the offset could be derived from a json query and stored as an attribute in pyportal.py.  (I'm way out of my league here.)

ladyada, 2019-03-18:
hmm - how would you ask Cpython for utc and local time?

makermelissa, 2019-03-18:
Couldn't you just temporarily set your timezone to something like "Etc/UTC" to get UTC?

CedarGroveStudios, 2019-03-18:
Here's how it works in the Python3 REPL: 
```
Jupyter QtConsole 4.3.1
Python 3.6.3 (v3.6.3:2c5fed8, Oct  3 2017, 17:26:49) [MSC v.1900 32 bit (Intel)]
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

>>> import time

>>> time.localtime()
Out[2]: time.struct_time(tm_year=2019, tm_mon=3, tm_mday=18, tm_hour=0, tm_min=3, tm_sec=26, tm_wday=0, tm_yday=77, tm_isdst=1)

>>> time.gmtime()
Out[3]: time.struct_time(tm_year=2019, tm_mon=3, tm_mday=18, tm_hour=7, tm_min=3, tm_sec=32, tm_wday=0, tm_yday=77, tm_isdst=0)

>>> time.gmtime()[3] - time.localtime()[3]
Out[7]: 7

```

CedarGroveStudios, 2019-03-18:
^^ I'll try that for now. Thanks @makermelissa!

ladyada, 2019-03-18:
sounds like we should add `gmtime()` to `time` at some point, wanna open up an issue on circuitpython for that? then we would be able to have two times!

CedarGroveStudios, 2019-03-18:
I'll do that! I'll close this issue since I think we have a potential workaround with @makermelissa 's suggestion.

ladyada, 2019-03-18:
yay!

CedarGroveStudios, 2019-03-18:
Done!  [Adafruit/circuitpython #1663](https://github.com/adafruit/circuitpython/issues/1663)

closed 2019-03-18 by CedarGroveStudios