adafruit/Adafruit_CircuitPython_PyPortal #17: Remove backward-compatible imports when 4.0.0 released

dhalbert, 2019-03-20:
Remove the `try`-`import` code that supports older versions of `displayio` et al when 4.0.0 is released as stable.

kevinjwalters, 2019-05-16:
On a related note, we have been discussing in discord the usefulness of this in https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/master/adafruit_pyportal.py#L58

```
try:
    from adafruit_display_text.text_area import TextArea  # pylint: disable=unused-import
    print("*** WARNING ***\nPlease update your library bundle to the latest 'adafruit_display_text' version as we've deprecated 'text_area' in favor of 'label'")  # pylint: disable=line-too-long
except ImportError:
from adafruit_display_text.Label import Label
```

The current behaviour for that is to print the warning which _cannot be seen_ as it's only on screen for about 1/4 of a second before the raster display blanks it. The import then has polluted the the code in some way where it blows up later on with a `NameError: name 'Label' is not defined. I'd say that should either be a "file exists" type test to avoid polluting `import` xor it should be a fatal error to a) make it visible and b) force user to address it.

(I had missed the step in instructions to delete the old files on **CIRCUITPY** in https://learn.adafruit.com/adabox011/updating-your-pyportal and based on forum discussion a small number of other users must have done the same: https://forums.adafruit.com/viewtopic.php?f=59&t=149573)

ladyada, 2019-05-17:
we're going to remove it soon when 4 is released which solves it, right?

kevinjwalters, 2019-05-17:
Yes, if you are referring to removing the `import from adafruit_display_text.text_area`.

PR #76 (2020-04-27): remove old style imports
I think this resolves #17 

Tested with https://learn.adafruit.com/pyportal-github-stars-trophy/code-pyportal-with-circuitpython on PyPortal with CP Versions 4.0.0, 5.2.0, and 5.3.0.rc0
closed 2020-04-27 by tannewt