adafruit/Adafruit_CircuitPython_PyPortal #21: PyPortal - Demo code: ValueError: syntax error in JSON (after 72 hours)

TheRealByron, 2019-04-04:
I believe I've encountered an error in the PyPoral Library that supports the code.py that retrieves Adafruit famous quotations. After an estimated 72 hours of "burn-in", I found the screen in text mode with a stacktrace-looking stream of text:
rieving data...Reply is OK! (beginning of that has scrolled off screen)
Couldn't parse JSON
The actual "text" value appears truncated, and there is no attribution.
The mentioned locations are:
   File code.py line 33 in <module>
   File code.py line 30 in <module>
   File adafruit_pyportal.py line 661 in fetch
   File adafruit_pyportal.py line 657 in fetch
   File adafruit_esp32spi/adafruit_esp32spi_requests.py line 110 in json
ValueError: syntax error in JSON

Code done running.Waiting for reload.

[image]

tannewt, 2019-04-04:
We should catch parse errors like this and retry a few times before raising an exception.

makermelissa, 2019-04-04:
I agree. This could be placed in a try block and handled better.

ladyada, 2019-04-04:
i commited a minor fix
https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/601
@TheRealByron wanna try that?

kattni, 2019-04-05:
@TheRealByron I have merged the fix. Please let us know if the updated code works for you.

ladyada, 2019-04-05:
closin' cause im pretty sure its solved - reopen if not!

closed 2019-04-05 by ladyada

CedarGroveStudios, 2019-04-06:
I've experienced a similar issue with the Weather Station tutorial code. I put this fix into my custom code and will test it.

ysiegel29, 2019-06-30:
Hello, I have had similar issues with a custom version of weather station demo code (using open weather). I could not solve the issue (or handle the issue), the code regularly crashed when using the json() method so could not finish the project.
After few months, I have tried again this week-end having installed latest lib and circuit python librairies (4.10 beta) but I have the exact same symptoms.

The URL for API seems to work just fine, the code usually work fine for few minutes to few hours and and then crash on a json() method.

> File "adafruit_esp32spi/adafruit_esp32spi_requests.py", line 116, in json
> ValueError: syntax error in JSON

The adafruit_esp32spi lib is the latest one.

Can you help? Am I doing something wrong or the above mentioned issue is not yet solved?
Thanks!!!

deshipu, 2019-06-30:
I think the issue is that the received json is being truncated — possibly due to a timeout or some other network problem. The best way to work around this is probably to catch this exception and retry.

ymauray, 2019-07-27:
Not sure if it's a timeout of a memory issue, but it's impossible to get a 5 day weather forecast from OpenWeatherMap due to that json being truncated.