adafruit/Adafruit_CircuitPython_PyPortal #22: Hang without timeout on adafruit.io image conversion call

mikerenfro, 2019-04-07:
@dhalbert requested that I add this here. [I'm experiencing intermittent hangs on my office door sign code](https://github.com/mikerenfro/iot-office-door-sign/issues/1), and at least one of those hangs happened while communicating with the adafruit.io image conversion service. The PyPortal never showed any evidence of a response, and no timeouts were triggered.

kevinjwalters, 2019-05-23:
What's size of bmp file for the second issue you mentioned? I've had a correct size file but with some sort of corruption of image data or bmp format with PyPortal and the standard NASA Image viewer: https://forums.adafruit.com/viewtopic.php?f=60&t=151923

mikerenfro, 2019-05-23:
320x240 if it went through the adafruit.io conversion service. 297x122 for the original PNG (or BMP, if I was doing my own conversion).

kevinjwalters, 2019-05-23:
Apologies, I meant file size! I was curious if the file was same as HTTP `Content-Length`.

mikerenfro, 2019-05-23:
Not certain when the file was coming from adafruit.io. When I had the converted BMP on one of my own servers, Content-Length always matched the file size. But at times, the PyPortal would stop downloading before Content-Length had been reached.

kattni, 2020-05-04:
This appears to be resolved by #23

closed 2020-05-04 by kattni