adafruit/Adafruit_CircuitPython_PyPortal #24: pyportal_simpletest.py is empty

caternuson, 2019-04-18:
Was it suppose to have something?

ladyada, 2019-04-18:
hmm i never wrote a simpletest. wanna put the bitcoin demo in?

caternuson, 2019-04-18:
Sure, can add something. You're thinking this?
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/PyPortal_Bitcoin

Would it be better to have something that doesn't rely on any assets / subfolders? Not sure what. Just thinking of trying to keep it to one file that needs to be worried about.

ladyada, 2019-04-18:
oh yeah hmm i guess it could just display on the console then?

caternuson, 2019-04-18:
Kind of boring, but yah. Grab some text and show it. Like maybe the same text test from here:
https://learn.adafruit.com/adafruit-pyportal/internet-connect#connect-to-wifi-17-5
That would verify wifi / internet / and display are working?

ladyada, 2019-04-18:
yah agree!

caternuson, 2019-04-18:
pr merged. version bump?

ladyada, 2019-04-18:
w0rd!

caternuson, 2019-04-18:
fixed with #25 
version bumped to 2.1.0

closin, yo.

closed 2019-04-18 by caternuson