adafruit/Adafruit_CircuitPython_PyPortal #30: show_QR method call fails

seantibor, 2019-05-02:
I'm trying to show a QR code on the screen when a touch is detected. Perhaps I'm doing something wrong?

Here's my code:

```python 
while True:
    p = ts.touch_point
    if p:
        print('{} was pressed'.format(p))
        light_on_time = time.monotonic()
        if not backlight:
            backlight = 1.0
            pyportal.set_backlight(1.0)
        elif p[1] < 240//3 * 2:
            current_image += 1
            if current_image == len(images):
                current_image = 0
            pyportal.set_background(cwd+images[current_image][0], images[current_image][1]) 
        else:
            pyportal.show_QR(b'http://bit.ly/2Y6zPu7')
    if backlight and (time.monotonic() - light_on_time) > backlight_timeout_seconds:
        backlight = False
        pyportal.set_backlight(0)

    time.sleep(0.05)

```

I am getting the following error message 

```
Traceback (most recent call last):
  File "code.py", line 51, in <module>
  File "adafruit_pyportal.py", line 845, in show_QR
ValueError: pixel value requires too many bits
```

caternuson, 2019-05-02:
That function looks like it hasn't been updated for latest `displayio` API. Another example:
https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/3da6b3c4756405cfa83d69e2cfee70eb4d9b6d0f/adafruit_pyportal.py#L849
So could stand a refresh.

I'm wondering if this could be done by simply drawing pixel to pixel the QR code raster, and then using the `scale` feature of `Group` to make it a larger size.

caternuson, 2019-05-03:
@seantibor Please try the 3.0.0 release of this library.

seantibor, 2019-05-04:
I've been running this all day and it's working consistently. This is really great!

closed 2019-05-04 by seantibor