adafruit/Adafruit_CircuitPython_PyPortal #38: Add new feature to allow changes/additions to the fetched and parsed JSON dict

kevinjwalters, 2019-05-29:
It's useful in some cases to be able to do some post processing on the returned JSON object. This could be achieved with an optional argument called something like `json_transforms` allowing the `PyPortal` object to be created with a function or a list of functions which run over the JSON `dict` object from the parsed HTTP response body. These could be allowed to modify or add values to the `dict` which could then be picked up by the usual `image_json_path` mechanism.

For the case of the NASA viewer code ([Adafruit Learn: PyPortal NASA Image of the Day Viewer](https://learn.adafruit.com/pyportal-nasa-image-of-the-day-viewer)) as there is no support for video at the moment it would be useful to be able to deal with the odd occasion when this returns a YouTube video URL and turn that into a URL with a single still image.

kevinjwalters, 2019-11-08:
https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/tree/3.1.3 is latest at the moment and does not have the `json_transform` enhancement.

kattni, 2020-05-04:
This has been resolved.

closed 2020-05-04 by kattni