adafruit/Adafruit_CircuitPython_PyPortal #40: Support multiple SSID and multiple passwords

kevinjwalters, 2019-06-04:
This is easily added by detecting if the SSID and password are not strings and iterating over them. 

> ATMakersBill: Hi folks, is there a way to list multiple SSIDs in the secrets.py file for the PyPortal and other WiFi boards?
> anecdata: I think secrets.py is just a convenient place to put private stuff. Could make WiFi credentials an iterable, and randomize or cycle through in your code.
> ATMakersBill: The PyPortal class has code for managing the secrets file - I'm asking if that has any ability to use multiples.
> kjw: @ATMakersBill What's the use case for that? Portability between different places?
> ATMakersBill: yes
> cater: i think it's pretty hard wired: https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/master/adafruit_pyportal.py#L633

A second use-case is to deal with changing passwords, two passwords for one SSID allows for a device to work near-seamlessly across password changes.

ladyada, 2019-06-04:
you could structure the json to have sub-arrays, but please add it to the esp wifi manager first

jedgarpark, 2020-08-14:
I'd love this. Ideally it would attempt to connect to the first SSID:password pair 3-4 times before moving on to the next one, since my highest signal strength AP often takes a few attempts to connect but it preferable to a different one out in my workshop when I'm inside my studio.

tannewt, 2020-08-17:
FYI, I intend on splitting WifiManager out into it's own repo so that it can be used for native wifi too.

closed 2020-08-31 by makermelissa