adafruit/Adafruit_CircuitPython_PyPortal #41: Building locally has issues on a macOS machine

lemonkey, 2019-07-08:
On a macOS machine with Xcode 10.2.1 installed, attempting to build from source by following the instructions on https://github.com/adafruit/Adafruit_CircuitPython_PyPortal#building-locally results in the following error:

```
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
../py/nlrx64.c:44:9: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
    #if NLR_OS_WINDOWS
        ^
../py/nlrx64.c:37:25: note: expanded from macro 'NLR_OS_WINDOWS'
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
                        ^
../py/nlrx64.c:44:9: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
../py/nlrx64.c:37:44: note: expanded from macro 'NLR_OS_WINDOWS'
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
                                           ^
../py/nlrx64.c:112:9: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
    #if NLR_OS_WINDOWS
        ^
../py/nlrx64.c:37:25: note: expanded from macro 'NLR_OS_WINDOWS'
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
                        ^
../py/nlrx64.c:112:9: error: macro expansion producing 'defined' has undefined behavior [-Werror,-Wexpansion-to-defined]
../py/nlrx64.c:37:44: note: expanded from macro 'NLR_OS_WINDOWS'
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
                                           ^
4 errors generated.
make: *** [build/py/nlrx64.o] Error 1
```

If you look at ./build_deps/circuitpython/py/nlrx64.c, on line 37 it has:

```
#define NLR_OS_WINDOWS (defined(_WIN32) || defined(__CYGWIN__))
```

On a Mac without Cygwin installed, neither _WIN32 nor __CYGWIN__ will be defined, which leads to the "macro expansion producing 'defined' has undefined behavior" error, failing the build.

If I replace the offending define with `#define NLR_OS_WINDOWS 0`, everything builds successfully.

However, if I try to install the compiled adafruit_pyportal.mpy file onto my PyPortal device using the newly built adafruit-circuitpython-pyportal-4.x-mpy-3.0.4 bundle, I get the incompatible .mpy error on soft boot. This doesn't happen with the same bundle version as downloaded from the official release.

Thoughts?

ladyada, 2019-07-08:
hiya why are you trying to build it locally? you want a new mpy?

lemonkey, 2019-07-09:
@ladyada I forked the repo in order to implement my own local Node.JS version of the image conversion service normally hosted on Adafruit.IO since I have local webcams that use the older style HTTP basic authentication for their image URLs.

ladyada, 2019-07-09:
hiya no need to do that, you can use mpycross here https://github.com/adafruit/circuitpython/releases/3.1.1/
or, on m4 boards, you can delete the old mpy file, you dont need mpy for m4/samd51 chips!

closed 2019-07-09 by ladyada

lemonkey, 2019-07-09:
@ladyada Oh I wasn't aware that mpy wasn't needed for the m4/samd51 chips since the latest auto-release only has the .mpy versions of the libraries, thanks!

lemonkey, 2019-07-09:
When copying over the latest version of the adafruit_pyportal.py file into my lib folder on the pyportal, on startup it complains about "no module named 'adafruit_touchscreen' since it in fact doesn't exist in the latest Adafruit_CircuitPython_Bundle that I originally installed (4.x).

lemonkey, 2019-07-09:
Nevermind, I had a storage space issue on the pyportal such that a few of the libraries failed to copy over.