adafruit/Adafruit_CircuitPython_PyPortal #46: recent update breaks PyPortal

jerryneedell, 2019-07-28:
It looks like #45 breaks this library on the PyPortal! It imports cursorcontrol which is imports gamepadshift which is not supported on the PyPortal.

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.0-45-gd99d3bd47-dirty on 2019-07-25; Adafruit PyPortal with samd51j20
>>> 
>>> 
>>> 
>>> import adafruit_pyportal
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_pyportal.py", line 60, in <module>
  File "adafruit_cursorcontrol/cursorcontrol_cursormanager.py", line 33, in <module>
ImportError: no module named 'gamepadshift'
>>> 
```

ladyada, 2019-07-28:
ok @brentru can take a look on monday - @jerryneedell if you want to try submitting a PR, that would be rad

jerryneedell, 2019-07-28:
I may have time later today to look into it. Will try.

jerryneedell, 2019-07-28:
I made a quick change and it now imports -- is this the way to go -? I just used the test it uses later to choose touchscreen or cursor.
```
diff --git a/adafruit_pyportal.py b/adafruit_pyportal.py
index b4f2668..e3bb826 100644
--- a/adafruit_pyportal.py
+++ b/adafruit_pyportal.py
@@ -56,8 +56,9 @@ from adafruit_esp32spi import adafruit_esp32spi, adafruit_esp32spi_wifimanager
 import adafruit_esp32spi.adafruit_esp32spi_requests as requests
 
 import adafruit_touchscreen
-from adafruit_cursorcontrol.cursorcontrol import Cursor
-from adafruit_cursorcontrol.cursorcontrol_cursormanager import CursorManager
+if hasattr(board, 'BUTTON_CLOCK'):
+    from adafruit_cursorcontrol.cursorcontrol import Cursor
+    from adafruit_cursorcontrol.cursorcontrol_cursormanager import CursorManager
 
 try:
     from adafruit_display_text.text_area import TextArea  # pylint: disable=unused-import
```

ladyada, 2019-07-28:
how about import cursor if there's no touchscreen?

jerryneedell, 2019-07-28:
ok - like this ```diff --git a/adafruit_pyportal.py b/adafruit_pyportal.py
index b4f2668..002747d 100644
--- a/adafruit_pyportal.py
+++ b/adafruit_pyportal.py
@@ -55,9 +55,11 @@ import neopixel
 from adafruit_esp32spi import adafruit_esp32spi, adafruit_esp32spi_wifimanager
 import adafruit_esp32spi.adafruit_esp32spi_requests as requests
 
-import adafruit_touchscreen
-from adafruit_cursorcontrol.cursorcontrol import Cursor
-from adafruit_cursorcontrol.cursorcontrol_cursormanager import CursorManager
+if hasattr(board, 'TOUCH_XL'):
+    import adafruit_touchscreen
+else
+    from adafruit_cursorcontrol.cursorcontrol import Cursor
+    from adafruit_cursorcontrol.cursorcontrol_cursormanager import CursorManager
 
 try:
     from adafruit_display_text.text_area import TextArea  # pylint: disable=unused-import
```

ladyada, 2019-07-28:
oooh ooh actually... if hasattr touch, import touch, if hasattr button_clock import cursormanager?

jerryneedell, 2019-07-28:
OK -np 
```
diff --git a/adafruit_pyportal.py b/adafruit_pyportal.py
index b4f2668..b9645c7 100644
--- a/adafruit_pyportal.py
+++ b/adafruit_pyportal.py
@@ -55,9 +55,11 @@ import neopixel
 from adafruit_esp32spi import adafruit_esp32spi, adafruit_esp32spi_wifimanager
 import adafruit_esp32spi.adafruit_esp32spi_requests as requests
 
-import adafruit_touchscreen
-from adafruit_cursorcontrol.cursorcontrol import Cursor
-from adafruit_cursorcontrol.cursorcontrol_cursormanager import CursorManager
+if hasattr(board, 'TOUCH_XL'):
+    import adafruit_touchscreen
+elif hasattr(board, 'BUTTON_CLOCK'):
+    from adafruit_cursorcontrol.cursorcontrol import Cursor
+    from adafruit_cursorcontrol.cursorcontrol_cursormanager import CursorManager
 
 try:
     from adafruit_display_text.text_area import TextArea  # pylint: disable=unused-import
```

but now and unrealted issue.  I see that pyportal_simpletest.py fails with:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.0-45-gd99d3bd47 on 2019-07-28; Adafruit PyPortal with samd51j20
>>> 
>>> import pyportal_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyportal_simpletest.py", line 10, in <module>
  File "adafruit_pyportal.py", line 173, in __init__
ValueError: TFT_BACKLIGHT in use
>>> 

```
:-(

ladyada, 2019-07-28:
oi i think maybe 5.0 changed stuff, wanna try 4.1?

jerryneedell, 2019-07-28:
OK -- I can look into that  - -should I put in the PR for the cursor/touch?

ladyada, 2019-07-28:
yez plz :)

PR #47 (2019-07-29): choose touch screen or cursor control
attempt to fix #46 
import either touchscreen or cursor control

minimally tested -- it does import now on a PyPortal
closed 2019-07-29 by brentru