adafruit/Adafruit_CircuitPython_PyPortal #53: wait_for_frame fails with CP 5.x

jerryneedell, 2019-09-29:
Does this call just have to be removed?

https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/master/adafruit_pyportal.py#L553

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.4-72-g7f1bc48a8 on 2019-09-20; Adafruit PyPortal with samd51j20
>>> 
>>> import pyportal_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pyportal_simpletest.py", line 10, in <module>
  File "adafruit_pyportal.py", line 245, in __init__
  File "adafruit_pyportal.py", line 243, in __init__
  File "adafruit_pyportal.py", line 553, in play_file
AttributeError: 'Display' object has no attribute 'wait_for_frame'
>>> 
>>> 
```

ladyada, 2019-09-29:
@tannewt is there any dox on the new screen drawing timing?

jerryneedell, 2019-09-29:
I removed line 553 and ran it with/without a pyportal_starup.wav on both Cp4.1.0 and CP5.0alpha-4  -- seems to work 
should I put in a PR?

tannewt, 2019-09-30:
@ladyada The only doc I know of is the API doc here: https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/Display.html#displayio.Display.refresh

Thanks @jerryneedell, should be ok to remove.

PR #54 (2019-09-30): remove call to wait_for_frame
fixes #53 
tested on pyportal under 4.1.0 and 5.x master -- works with and with or pyportal_startup.wav  file.
closed 2019-09-30 by tannewt