adafruit/Adafruit_CircuitPython_PyPortal #57: `get_local_time` doesn't properly check the http response status code before attempting to parse the time

theacodes, 2019-11-27:
https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/blob/81c82035649007bbccb7d0ec75ce192f3b2ca058/adafruit_pyportal.py#L618

If the request is an error, such as `404` for `User not found` ([example url](https://io.adafruit.com/api/v2/idontexistipromise/integrations/time/strftime?tz=America/New_York&fmt=%25Y-%25m-%25d+%25H%3A%25M%3A%25S.%25L+%25j+%25u+%25z+%25Z)), then this code will attempt to parse the response into a datetime anyway.

It *should* makes sure the status code is `200` before attempting to parse. If it's not, it should throw an exception and include the response text in the exception to help with debugging. As it stands, this will raise a very unhelpful error:

```
Traceback (most recent call last):
  File "code.py", line 44, in <module>
  File "adafruit_pyportal.py", line 629, in get_local_time
  File "adafruit_pyportal.py", line 625, in get_local_time
ValueError: invalid syntax for integer with base 10 
```

This would be an excellent first-time issue for someone.

ladyada, 2019-11-27:
good idea - thanks for spotting this one @theacodes 
@brentru might also be a good one for you :)

brentru, 2019-11-27:
@ladyada  I'll leave it for a new contributor within the next 15 days as people's holiday purchases roll in. I'll patch/release by then if it doesn't get picked up.

brentru, 2020-01-28:
Addressed in https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/pull/63.

Merged into `master` and released in `v3.1.8`.

closed 2020-01-28 by brentru