adafruit/Adafruit_CircuitPython_PyPortal #67: AttributeError: 'Display' object has no attribute 'refresh_soon'

avondohren, 2020-03-05:
Hello. I apologize if this issue has already been reported or fixed. I'm just getting started with the PyPortal and I'm getting the following error when trying to upgrade to CircuitPython 5.0.0.
```
Set background to  /pyportal_startup.bmp
Traceback (most recent call last):
  File "code.py", line 23, in <module>
  File "adafruit_pyportal.py", line 214, in __init__
  File "adafruit_pyportal.py", line 207, in __init__
  File "adafruit_pyportal.py", line 386, in set_background
AttributeError: 'Display' object has no attribute 'refresh_soon'
```

This is a brand new PyPython that I ordered from DigiKey, and I downloaded the latest CircuitPython .UF2 here https://circuitpython.org/board/pyportal/

Is there a way to edit the `adafruit_pyportal.py` to remove the references to `refresh_soon`?

In doing some research, I see in the release notes that the `refresh_soon` option was removed in version 5.0.0. (https://github.com/adafruit/circuitpython/releases/tag/5.0.0)

For now I'll roll back to the version that came pre-installed on my device. But I'd like to install the latest version.

Thanks for your time and consideration. 

Andy

avondohren, 2020-03-05:
Okay, after digging around some more I think I may have gotten 5.0.0 working.

After looking at the latest CircuitPython release bundles here: https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20200304

I downloaded the v5.x bundle - adafruit-circuitpython-bundle-5.x-mpy-20200304.zip

After unpacking the zip, I re-installed 5.0.0 then updated the following libraries via drag-and-drop with USB:
- adafruit_pyportal.mpy (updated)
- adafruit_requests.mpy (new)
- adafruit_io (new)

Once those libraries were updated/added the PyPortal was able to successfully boot and run my program! 🙌

avondohren, 2020-03-05:
Well, I may have spoken too soon. Although the program does run, it's not able to complete the `pyportal.fetch()` command. It gets the following error:
```
Retrieving data...Traceback (most recent call last):
  File "code.py", line 34, in <module>
  File "code.py", line 31, in <module>
  File "adafruit_pyportal.py", line 801, in fetch
  File "adafruit_requests.py", line 280, in get
  File "adafruit_requests.py", line 242, in request
  File "adafruit_requests.py", line 195, in request
AttributeError: 'socket' object has no attribute 'send'
```

avondohren, 2020-03-05:
Okay. I went through the `lib` folder and updated all of the libraries to the 5.x equivalent from the `adafruit-circuitpython-bundle-5.x-mpy-20200304.zip` download previously mentioned. Once I did that my script not only loaded, but was able to connect to the API and update the display! 🙌 

I think there needs to be some work in order to ensure that when updating to 5.0.0 the libraries also get updated or that the user is pointed to update their libraries. This was obviously very confusing for me tonight.

This all started because I wanted to loop through a list of stock symbols so that every refresh would pull back a different stock price. The ability to add a `refresh_url=` param to the `.fetch` function required a more recent version of the PyPortal library, but getting to the point where it worked took a long time and I almost gave up 3 times. 

Let me know if there is anything I can do to contribute back to the project with what I learned. Like I said, I'm just getting started with this board, but I'm hoping to make a decent little personal dashboard out of it so I may find opportunities to keep improving it in the future.

FoamyGuy, 2020-03-05:
@avondohren Check out the [CircUp Tool](https://learn.adafruit.com/keep-your-circuitpython-libraries-on-devices-up-to-date-with-circup/overview) If you are familiar with NPM, pip, or any other package management type utilities this is a similar type of tool made for Circuit Python libraries. It gives you an easy to use command line interface to check for and update libraries on your Circuit Python devices.

avondohren, 2020-03-05:
@FoamyGuy Thanks for the tip. I didn't know that tool existed. Would have saved me a bunch of time!

jwindberg, 2020-03-17:
The instructions for the pyportal at: 
https://learn.adafruit.com/adabox011/updating-your-pyportal
Will kill your pyportal. The update to 5.0.0 and the current examples are not compatible with each other.

jwindberg, 2020-03-17:
circup did not help, just changed the error message
I get no module named adafruit_requests

caternuson, 2020-03-17:
"Killing" implies a hardware bricking, which should not happen. Having libraries get out of sync with the main firmware can happen though. Updating one without the other is a common way for this to happen. I think circup can only check status of currently installed libraries. It can't determine that you are missing a library. The [requests library](https://github.com/adafruit/Adafruit_CircuitPython_Requests) is in the [bundle](https://circuitpython.org/libraries).

If you update **both** the firmware and the libraries, and make sure all necessary libraries are installed, is there still an issue?

jwindberg, 2020-03-17:
point
found adafruit_results.py, then had to copy adaruit_io folder.
Now I see 
File adafruit_requests.py line 195
AttributeError: 'socket' object has no attribute 'send'
trying another cirqup update
and there it goes, nice.
Need to get someone to update the example zip for pyportal.
If someone, who is not that familiar with python, like me, wants to get started, the example code ought work out of the box.

kattni, 2020-05-04:
Closing this issue. Looking into updating or better documenting the PyPortal Adabox 011 update page.

closed 2020-05-04 by kattni