adafruit/Adafruit_CircuitPython_PyPortal #72: Default param for set_font

cpirnat, 2020-04-06:
Currently set_font on PyPortal class is set to None. Is there a chance we could utilize termalio font for a default font?

siddacious, 2020-04-06:
What is the issue you are trying to solve? Please give a code example of what you are trying to do that isn't working

cpirnat, 2020-04-06:
This would assist the helper library in be able to write text to display without the need of manually importing a font by the end user.

siddacious, 2020-04-06:
That seems useful; can you give a mock example of how you would like it to work? What specific code do you have in mind to write that won't work as is?

`text_font` takes a string path to a font file which `terminalio.FONT` is not, so it would not be as simple as adding a default value.

dherrada, 2020-07-01:
hi @cpirnat Are you still planning on working on this?

makermelissa, 2020-10-08:
I implemented this in the MatrixPortal library, so adding it to here should be pretty straightforward. In fact, I may have done this already on a PyPortal and have it floating around somewhere. I think there were only a handful of places special cases needed to be addressed.

makermelissa, 2020-10-08:
Ah yes, I found it. :)

PR #86 (2020-10-23): Added terminalio font support and scaling
Fixes #72.
closed 2020-10-23 by makermelissa