adafruit/Adafruit_CircuitPython_PyPortal #89: It would be nice to have a few more examples

FoamyGuy, 2020-10-10:
Right now the only example is simpletest which (rightly so) only scratches the surface of what this library can do. 

It would be great to get a few more examples built up that show the most basic ways to handle and show text, JSON data, and images on the screen. 

I can work on adding some of these, but this could also server as a good first issue if anyone is interested.

brentru, 2020-10-13:
@FoamyGuy  The New New New! Viewer (https://learn.adafruit.com/pyportal-new-new-new-product-viewer/code-the-new-new-new-viewer-in-circuitpython) would be a good example sketch which touches on all of the criteria you listed.

jposada202020, 2021-04-21:
@FoamyGuy 
I see these guides as interesting choices. you are thinking to add them to the examples code, or to the library API read_the_docs, tutorial category?

https://learn.adafruit.com/iot-pyportal-data-logger-adafruitio-circuitpython
https://learn.adafruit.com/pyportal-smart-thermometer-with-analog-devices-adt7410-adafruit-io-and-circuitpython
https://learn.adafruit.com/pyportal-iss-tracker

Let me know thanks.

FoamyGuy, 2021-04-21:
@jposada202020 My intention originally was for the examples folder inside of the repo. I tend to look there first when I am trying to learn how to use a library.

I think the learn guide projects are really nice but might have a bit too much going on within them to be included as example scripts in this repo. Ideally the examples used for this should not require much if any extra hardware / printed parts.

I believe I was thinking of things like pulling basic data from some API online and showing it to the user on the screen. Nothing super fancy necessarily, just the functionality of grabbing that data and showing.

jposada202020, 2021-04-21:
@FoamyGuy Gotcha :)

makermelissa, 2021-04-22:
Fixed via #111.

closed 2021-04-22 by makermelissa