adafruit/Adafruit_CircuitPython_PyPortal #94: push_to_io

HerbertUnterberger, 2020-11-10:
with the version 6 its a first that I can reach the image converter from Vienna.
thx a lot to all of you working on requests and PyPortal

But here:

pyportal.push_to_io goes straight in a loop with the message:

An error occured, retrying! 1 - Sending request failed
An error occured, retrying! 1 - Sending request failed
An error occured, retrying! 1 - Sending request failed
An error occured, retrying! 1 - Sending request failed

Herbert

tannewt, 2020-11-10:
Hi! Please make sure you have the latest ESP32FSPI and Requests libraries.

HerbertUnterberger, 2020-11-10:
Requests is newest

ESP32FSPI is from Bundle

I will copy it tomorrow

thx for advise

HerbertUnterberger, 2020-11-11:
Hi!
ESP32FSP 3.5.3 did not change the behavior still a retrying loop.

tannewt, 2020-11-12:
Thanks! Could you post or link to the full code example? I'll take a look tomorrow.

HerbertUnterberger, 2020-11-12:
Hi, Scott

please find the code in the in the link

https://www.dropbox.com/s/sm6i07senqn30gc/code.py?dl=0

It is more ore less like the Amazon Dash Button, but with pictures and text.

Herbert

makermelissa, 2020-11-12:
@brentru do you think this is related to the recent changes to Adafruit IO?

HerbertUnterberger, 2020-12-01:
@makermelissa brentru commented 11 days ago
Update examples for compatibility with the ESP32-S2 and updated requests module.

Remove dependency for WiFiManager (https://github.com/adafruit/Adafruit_CircuitPython_AdafruitIO/blob/master/adafruit_io/adafruit_io.py#L473) and move to a http_api.set_socket(socket) interface instead to match requests.

socket.set_interface(esp)
requests.set_socket(socket)
```

could that be reason?

makermelissa, 2020-12-07:
Possibly. I plan on doing a refactor of this library soonish to use a unified base which should fix this issue. If it doesn't happen as soon as I think, I may just do a quick patch to fix this.

brentru, 2020-12-09:
@makermelissa  This **is** related to the change in the Adafruit IO library, the lib's constructor doesn't take in WiFiManager anymore and instead takes in a socket interface.

makermelissa, 2021-01-09:
I now have a refactor PR (#99) in which should fix this once merged.

makermelissa, 2021-01-13:
Ok, the refactor has been released which should fix this. Please update libraries and try again. Thanks.

closed 2021-01-13 by makermelissa