adafruit/Adafruit_CircuitPython_PyPortal #95: I copied pyportal_simpletest.py to cord.py and got an error.

NekomuraHana, 2020-11-15:
When I copied the code of pyportal_simpletest.py to code.py and made it work, I got the following error.

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Set background to  /pyportal_startup.bmp
ESP firmware: bytearray(b'1.2.2\x00')
Connecting to AP home-AP-n-24
Set background to  0
No SD card found: timeout waiting for v2 card
Traceback (most recent call last):
  File "code.py", line 10, in <module>
  File "adafruit_pyportal.py", line 378, in __init__
TypeError: 'NoneType' object is not subscriptable

Press any key to enter the REPL. Use CTRL-D to reload.
```

How can I resolve this error?
I'm not familiar with python so I don't know if it's a bug.
I want you to help me.
The version of CircuitPython is 5.3.1.
Nice to meet you.

FoamyGuy, 2020-11-15:
This was fixed in a recent new version of the PyPortal library. Get the latest 5.X bundle from here: https://circuitpython.org/libraries and copy the newest pyportal library mpy file into the `lib` folder on your device. This page has more info too: https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-libraries

FoamyGuy, 2020-11-17:
@NekomuraHana I'm going to close this for now. We can re-open it if you are still having trouble.

closed 2020-11-17 by FoamyGuy