adafruit/Adafruit_CircuitPython_PyPortal #114: PyPortal mqtt integration with UI example

oldblackcrow, 2021-05-04:
Hello all!

For a few months, I've been trying to integrate mqtt with the PyPortal UI example (https://learn.adafruit.com/making-a-pyportal-user-interface-displayio). @makermelissa and @brubell have been wonderful in getting mqtt to work on the PyPortal. Believing I had a clear path, the issue is that the workaround code on this page (https://forums.adafruit.com/viewtopic.php?f=56&t=177198&start=15) does not use the pyportal = PyPortal() protocol. I thought that solved my problem, but when I integrate that into the UI example, it fails because the UI **does** use pyportal = PyPortal(). 

On the page, Make It Your Own (https://learn.adafruit.com/making-a-pyportal-user-interface-displayio/make-it-your-own), it challenges us to integrate the mqtt into the UI... which is where I started on this long journey.

I don't know if there are plans to remedy this or if it's better to pull that Make It Your Own page down. I would LOVE to see this resolved though, as my project was designed to rely on this. Thank you for your time and effort and please contact me with any questions.

makermelissa, 2021-05-04:
Currently we have an initial MQTT implementation in the FunHouse library, which is very similar. Ideally it could be moved to PortalBase which would add the functionality to the PyPortal, MatrixPortal, and MagTag devices as well. However, the libraries are quite large at this point and doing so may cause memory issues, which is why I haven't just put it there.

oldblackcrow, 2021-05-04:
I can understand that for sure... 

I am intrigued, however, as the PyPortal does have the SD card... or are you talking about active RAM?

makermelissa, 2021-05-04:
Active RAM

oldblackcrow, 2021-05-06:
Ok... Thanks to @Neradoc#2644 on Discord, this issue has been solved... At least for me. His original code is located here: https://gist.github.com/Neradoc/0093bac7d0ec5608f4fa7918e4950f1d

The working modified code for the PyPortal is:

`import` time
import random
from adafruit_pyportal import PyPortal

pyportal = PyPortal()
pyportal.network.connect()

key = "rx"
latest_id = "rx"

while True:
    feed = pyportal.get_io_feed(key)
    client = pyportal.network._get_io_client()
    last = client.receive_data(key)
    the_id = last['id']
    if the_id != latest_id:
        print("New value:",last['value'])
        latest_id = the_id
    time.sleep(5)``